ARMmbed/uvisor-helloworld #20: uvisor-helloworld could not work on STM32F429I-Discovery

georgekang, 2015-10-01:
When testing uvisor-helloworld on STM32F429I-Discovery, only red LED would blink 
with error pattern 2, SANITY_CHECK_FAILED, and CPU would finally in an infinite 
loop from address 0x0800123e to 0x0800124c.

jserv, 2015-10-05:
@georgekang : It is weird. Can you attach the corresponding log while uvisor falls into infinite loop on STM32F429i-Discovery?

georgekang, 2015-10-06:
@jserv : uvisor-helloworld would fall into infinite loop when Uvisor was fetched by yotta from https://github.com/ARMmbed/uvisor-lib.  Because it was a binary release and there was no 
output log except  LED blinking.  

Besides, I find some problems when Uvisor tried to start the app program(app_start).
  (1). Accessing SYSCFG registers would cause MPU fault even through these registers were
        declared as an acl item in box (source/main-hw.h). 
```
#define MAIN_ACL(acl_list_name)                           \
    static const UvisorBoxAclItem g_main_acl[] = {        \
        {TIM2,   sizeof(*TIM2),   UVISOR_TACLDEF_PERIPH}, \
        {TIM5,   sizeof(*TIM5),   UVISOR_TACLDEF_PERIPH}, \
        {GPIOA,  sizeof(*GPIOA),  UVISOR_TACLDEF_PERIPH}, \
        {GPIOG,  sizeof(*GPIOG),  UVISOR_TACLDEF_PERIPH}, \
        /* FIXME: secure RCC/EXTI/SYSCFG */               \
        {RCC,    sizeof(*RCC),    UVISOR_TACLDEF_PERIPH}, \
        {EXTI,   sizeof(*EXTI),   UVISOR_TACLDEF_PERIPH}, \
        {SYSCFG, sizeof(*SYSCFG), UVISOR_TACLDEF_PERIPH}, \
    }
``` 

  (2).   Before real_main was executed, HAL_Init() was called to restart HAL driver.  
          However, this function would use some registers, such like FLASH and SCB, which
          were not declared in main acl list and bus fault exception would be triggered because
          these registers were used in unprivileged mode.  
   
  (3).  minar::Scheduler::start () used HAL_PWR_EnterSLEEPMode() to make core in sleeping.
         However, core would fall into exception mode with bus fault like (2) because registers in SCB
         were accessed in HAL_PWR_EnterSLEEPMode().

meriac, 2015-10-06:
We're experiencing a related problem - I expect it will fix the previous one. When accessing the flash access control register at 0x40023c00, an exception is generated - but the MMFAR register is inconsistent and points to the previous fault.

georgekang, 2015-10-06:
According to B.3.5.1, Relation of the MPU to the system memory map, in ARMv7-M Architecture Reference Manual, the MPU is restricted in how it can change the default memory map attributes associated with System space, that is, for addresses 0xE0000000 and higher. 

MPU could not change the permissions of System space registers like SCB, NVIC, and SysTick and we should avoid using these registers in unprivileged mode.   Currently, these registers are used in `mbed_hal_init() --> HAL_Init()` and `minar::Scheduler::start () --> HAL_PWR_EnterSLEEPMode()` and cause bus fault exception.

meriac, 2015-10-06:
We are currently developing the feature "register level security" that takes care of that - without adding custom APIs for each of these user-cases:

https://github.com/ARMmbed/uvisor/issues/38

That feature will also enable bit level access restrictions to registers.

georgekang, 2015-10-07:
Thank @meriac to reply my problems.  

In conclusion, some registers are critical and can not be used in unprivileged mode directly.  The new feature "register level security" will be developed in Uvisor to provide restricted access for these registers to user box which is executed in unprivileged mode. 

I look forward for further developing of this new feature.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-838

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1022

AlessandroA, 2015-10-28:
Fixed by:
https://github.com/ARMmbed/mbed-hal-st-stm32cubef4/pull/8
https://github.com/ARMmbed/cmsis-core-stm32f4/pull/16
https://github.com/ARMmbed/cmsis-core-stm32f429xi/pull/10
https://github.com/ARMmbed/uvisor/pull/83
https://github.com/ARMmbed/uvisor/pull/81

closed 2015-10-28 by AlessandroA