ARMmbed/uvisor-helloworld #24: The blue LED does not blink on K64F in debug mode

AlessandroA, 2015-10-06:
When uVisor debug messages are enabled, they are so many that they prevent minar from meeting deadlines for scheduled events (blinking the blue LED). This has been observed on K64F but could affect STM32F4 as well.

Since debug messages come from high-priority interrupts (system exceptions, mainly), they always overwrite the event handler priority.

We could use yotta config symbols to conditionally define events timeouts depending on the uVisor mode (release vs debug)

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-837

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1021

meriac, 2015-10-31:
fixed https://github.com/ARMmbed/uvisor/commit/5dba89284b6e343988bdbd41600a0fa23bc500ad

closed 2015-10-31 by meriac