ARMmbed/uvisor-helloworld #46: The build fails for non-K64F and non-STM32 targets

Patater, 2016-01-06:
When you attempt to build for targets that aren't TARGET_LIKE_FRDM_K64F, and aren't TARGET_LIKE_STM32F429I_DISCO, the following error is generated and the build fails.

```
        source/btn.cpp: In function 'void btn_init()':
        source/btn.cpp:39:27: error: no matching function for call to 'mbed::InterruptIn::mode(PinName)'
             btn.mode(MAIN_BTN_PUPD);
                                   ^
        source/btn.cpp:39:27: note: candidate is:
        In file included from yotta_modules/mbed-drivers/mbed-drivers/mbed.h:59:0,
                         from source/btn.cpp:17:
        yotta_modules/mbed-drivers/mbed-drivers/InterruptIn.h:109:10: note: void mbed::InterruptIn::mode(PinMode)
             void mode(PinMode pull);
                  ^
        yotta_modules/mbed-drivers/mbed-drivers/InterruptIn.h:109:10: note:   no known conversion for argument 1 from 'PinName' to 'PinMode'
```
    
The reason for this is because `NC` is defined as a `PinName`, not a `PinMode`. However, if you replace `NC` with a `PinMode`, a different error is generated and the build still fails.

```
    source/main.cpp:33:24: error: no matching function for call to 'mbed::DigitalOut::DigitalOut(PinMode)'
     DigitalOut led(MAIN_LED);
                            ^
    source/main.cpp:33:24: note: candidates are:
    In file included from yotta_modules/mbed-drivers/mbed-drivers/mbed.h:39:0,
                     from source/main.cpp:17:
    yotta_modules/mbed-drivers/mbed-drivers/DigitalOut.h:57:5: note: mbed::DigitalOut::DigitalOut(PinName, int)
         DigitalOut(PinName pin, int value) : gpio() {
         ^
    yotta_modules/mbed-drivers/mbed-drivers/DigitalOut.h:57:5: note:   candidate expects 2 arguments, 1 provided
    yotta_modules/mbed-drivers/mbed-drivers/DigitalOut.h:48:5: note: mbed::DigitalOut::DigitalOut(PinName)
         DigitalOut(PinName pin) : gpio() {
         ^
    yotta_modules/mbed-drivers/mbed-drivers/DigitalOut.h:48:5: note:   no known conversion for argument 1 from 'PinMode' to 'PinName'
    yotta_modules/mbed-drivers/mbed-drivers/DigitalOut.h:41:7: note: mbed::DigitalOut::DigitalOut(const mbed::DigitalOut&)
     class DigitalOut {
           ^
    yotta_modules/mbed-drivers/mbed-drivers/DigitalOut.h:41:7: note:   no known conversion for argument 1 from 'PinMode' to 'const mbed::DigitalOut&'

```

AlessandroA, 2016-01-25:
@Patater Which target did you use for this? I tried to reproduce the second error you mentioned but got a successful compilation (after changing `NC` to `PullDown`)

Patater, 2016-01-25:
I was attempting to target NRF51.

AlessandroA, 2016-01-25:
I just tried building with the mkit-gcc target (which is an NRF51 target) and it does not show the error. It could be that it was not related to the pre-processor macros and has been fixed since then. I am sending the fix for the `PinMode`. Could you please test it as well?

Patater, 2016-01-25:
Does your fix work for nrf51dk-gcc?

AlessandroA, 2016-01-25:
Yes it builds fine

AlessandroA, 2016-01-25:
(actually wait, I tested on nordic-nrf51822-gcc)

Patater, 2016-01-25:
Super.

AlessandroA, 2016-01-25:
Fixed in https://github.com/ARMmbed/uvisor-helloworld/pull/50

closed 2016-01-25 by AlessandroA