ARMmbed/uvisor-helloworld #48: CHALLENGE_SIZE leakage via timing attack

avri-schneider, 2016-01-24:
`__verify_secret()` accepts a `len` parameter which is verified, before the comparison with `secure_compare()`, to be equal to `CHALLENGE_SIZE`.

If verification fails, constant time comparison does not take place, and function returns immediately.

If `CHALLENGE_SIZE` is known by an attacker, why the need for the `len` parameter?
If `CHALLENGE_SIZE` is not known by an attacker, `CHALLENGE_SIZE` is leaked via timing attack.

meriac, 2016-01-26:
@avri-schneider @AlessandroA I see the "len" parameter as a safety feature. I don't like API's where the called target makes assumptions about the size of the provided memory. In that sense I decided to have both the caller and the callee state their idea of the buffer length and to only continue once both ideas of "len" match up.

As a callee determines the length of the comparison I claim that the information-leak for the caller is not measurable.

avri-schneider, 2016-01-26:
@meriac @AlessandroA

I assume by *caller* and *callee* you refer to *provider of untrusted value to be checked* and *holder of trusted expected secret*, respectively.

> I don't like API's where the called target makes assumptions about the size of the provided memory

You are passing two pointers, and already make an assumption that both are the same length, by passing a single `len` parameter.

> I decided to have both the caller and the callee state their idea of the buffer length and to only continue once both ideas of "len" match up

In that case you leak length of the secret held by *callee* to *caller*.

> As a callee determines the length of the comparison I claim that the advantage of the attacker through information-leakage for the caller is negligible compared to the effort of guessing the target string right.

This could be true in some situations, for example systems enforcing a minimum non-bruteforcable secret length, but false in other situations, where a small, bruteforcable length secrect is possible. This may allow attacker to find accounts holding bruteforcable secrets, and only attack them.

>  I claim that the advantage of the attacker through information-leakage for the caller is negligible compared to the effort of guessing the target string right.

Why should we leak *any* information to an attacker?

Since this code may be interperted as reference implementation for securely verifying a secret in constant time, care should be taken to clearly document the assumptions and cases where such code will not provide adequate security and should not be used.

In general, I think that code that claims to provide security features (such as run in constant time to protect against timing attacks) it is important to verify the security features are provided in all possible use-cases (i.e. no side-channel leakage via timing attacks are possible). Security assumptions must be clearly stated, otherwise developers may [will] take the code verbatim and apply it is contexs where the claimed security assurances are not provided.

meriac, 2016-01-27:
@avri-schneider The intention of the hello-world is to be the smallest possible security-hello-world example for demonstrating what uvisor actually protects against. It provides a clear indication for breaking the complex uvisor security model:

"When I can access or guess the secret as an internal attacker, uvisor is no good"

I claim that uvisor-helloworld AS-IS holds up to that promise. I see how I would fix the above timing attack - I don't see though how it can be done without complicating the above example and without additional protection in this specific instance.

Would it satisfy you to put a clear comment on top the comparison function that highlights that the secret length is not protected? @AlessandroA @Patater

meriac, 2016-01-27:
(merged comment, closed issue)

closed 2016-01-27 by meriac

avri-schneider, 2016-02-03:
@meriac @AlessandroA @Patater @armask @leo-dor

Sorry for the late response, was out-of-office this past week.

The comment you added is really nice and negates the need for a proper secure function being used in this demo app. Nevertheless, I am a bit uncomfortable with the choice of _secure_compare_ as a function name, and with the actual decision of implementing security functions as part of this demo app.

If you ask me, I would say the proper way to fixing this issue would be to provide library functions as part of mbed-os, for doing any operations having security implications, along with proper documentation and demo apps, such as this one, explaining the proper use of their APIs.

For example, this functions could live in a side-channel-protection library, included by uvisor-helloworld. This would serve both the purpose of demonstrating uvisor's complex security model, as well as teaching users how to properly use our secure APIs.

IMHO, I'd prefer we don't make users get accustomed to writing their own security functions, and I'm afraid this approach might lead them in that direction. Your thoughts?

meriac, 2016-02-10:
@avri-schneider Agreed on the need for such a "abstract security library" - i believe @sbutcher-arm is working on a generic high layer library that provides simplified high level security function like "hash this thing" - a secure compare would nicely fit into that. Please make a case for that inside ARM - I am happy to support it. I can imagine also other functions like "secure string compare" / "secure memset/memzero"  living there, too.

Once we have that, I am happy to modify the uvisor_hello_world code to make use of it.

avri-schneider, 2016-02-11:
@meriac Thanks. Will open a JIRA case.

We might also be able to use libsodium's [sodium_memcmp()](https://download.libsodium.org/doc/helpers/index.html)...

[libsodium](https://download.libsodium.org/doc/index.html) is released under [ISC license](https://en.wikipedia.org/wiki/ISC_license)