ARMmbed/uvisor-helloworld #53: How to keep Serial instance with arguments in BOX context

JaredCJR, 2016-02-01:
Hi,
I would like to create a BOX that need to use USART3 in STM32F429i-Discovery.
However,I am confused about how to keep an `Serial` instance in the secure BOX.

Why I want to do this is because I found that if the `Serial` instance is declared outside the BOX_Context,the background ACLs of the main box still needs the permission to access USART3.

`Serial mri_port(PB_10,PB_11); placed in struct` is invalid syntax in C++, so I am wondering how to keep the `Serial` instance secret with arguments.
 

```C
#define BOX_ACL(acl_list_name)                      \                                                                                                              
    static const UvisorBoxAclItem acl_list_name[] = {     \
        {GPIOB,  sizeof(*GPIOB),  UVISOR_TACLDEF_PERIPH}, \
        {RCC,    sizeof(*RCC),    UVISOR_TACLDEF_PERIPH}, \
        {USART3, sizeof(*USART3), UVISOR_TACLDEF_PERIPH}, \
    }

typedef struct {
    Serial mri_port(PB_10,PB_11);//USART3 pins in STM32F429
    uint32_t val;
} BOX_Context;

/* create ACLs for secret data section */
BOX_ACL(g_new_box_acl);                                                                                                                                          

/* configure secure box compartment and reserve box context */
UVISOR_BOX_CONFIG(new_box, g_new_box_acl, UVISOR_BOX_STACK_SIZE, BOX_Context);

```
Thanks.

meriac, 2016-02-02:
Hi @JaredCJR,

in the old version of uvisor-helloworld I created a similar example:

https://github.com/ARMmbed/uvisor-helloworld/blob/d002b0ac999ef53b3b543479e6cf2c0322ef2eff/source/box_secure_print.cpp

The trick is to use to use the "placement new" operator in C++ to construct an object at a specific location. Back then I had the conclusion that the serial object is insecure as it currently allocates memory in the main box - please use RawSerial instead.

closed 2016-02-02 by meriac

meriac, 2016-02-02:
(see line https://github.com/ARMmbed/uvisor-helloworld/blob/d002b0ac999ef53b3b543479e6cf2c0322ef2eff/source/box_secure_print.cpp#L56)

JaredCJR, 2016-02-02:
Thaks!
That is what I want!